timvideos/HDMI2USB-mode-switch #91: Add a "--version" to download-firmware script.

mithro, 2018-01-20:


Nancy-Chauhan, 2018-03-19:
@mithro Sir I have analysed that HDMI2USB-mode-switch/bin/download-prebuilt-firmware.py is responsible for downloading the firmware script and further analysed that adding --version to parseargs() can help us to solve this issue ? I want to ask you whether I am thinking in right direction or not ? can you provide me further detail about this issue . I will be thankful to you for this

CarlFK, 2018-03-19:
Nancy, 
yes, please do this.

we should probably use the same 'framework' as the mode-swtich command, which is here:

https://github.com/timvideos/HDMI2USB-mode-switch/blob/master/hdmi2usb/modeswitch/cli.py#L275
https://github.com/timvideos/HDMI2USB-mode-switch/blob/master/hdmi2usb/modeswitch/_version.py

also, maybe we should move download-firmware.py into this dir.

shaharyar-shamshi, 2018-03-19:
@CarlFK sir As I analyse the issue we have to just add latest version (which is to be downloaded  and make change in parseargs() as I have done in line no. 93 (https://paste.ubuntu.com/p/wvxkbHQPdS/) and make necessary change in make corresponding change in image_url please correct me if I am wrong . so that I can open pull request for it soon.

<blockquote><div><strong><a href="https://paste.ubuntu.com/p/wvxkbHQPdS/">Ubuntu Pastebin</a></strong></div></blockquote>

Nancy-Chauhan, 2018-03-19:
@CarlFK sir As I analyse the issue we have to just add latest version (which is to be downloaded and make change in parseargs() as I have done in line no. 93 (https://paste.ubuntu.com/p/wvxkbHQPdS/) and make necessary change in make corresponding change in image_url please correct me if I am wrong . so that I can open pull request for it soon.

https://paste.ubuntu.com/p/wvxkbHQPdS/

CarlFK, 2018-03-19:
"--version" the version of the download script, not the version of firmware
available.

I am going to be somewhat unavailable for the next 5 days, I suggest you
submit a pull request sooner and will have a look.