timvideos/HDMI2USB-mode-switch #106: assert len(exart_uarts) == len(atlys_boards)  (sometimes)

CarlFK, 2019-06-27:
most of the time this happens:
```
pi@oppi:~/prebuilt $ atlys-off-on.sh 

pi@oppi:~/prebuilt $ ~/HDMI2USB-mode-switch/venv/bin/hdmi2usb-mode-switch -v --by-type atlys --load-fx2-firmware hdmi2usb.hex
WARNING:root:unbind-helper not found, will have to run as root!
 Found exart-uarts at [LsusbDevice(04e2:1410:0003 /dev/bus/usb/001/015)] associating with Atlys at [Board(dev=LsusbDevice(1443:0007:0000 /dev/bus/usb/001/014), type='atlys', state='unconfigured')]
Digilent Atlys in 'unconfigured' mode at /dev/bus/usb/001/014
 /sys/bus/usb/devices/1-1.3.1
 /sys/bus/usb/devices/1-1.3.1:1.0
 Serial port at /dev/ttyVIZ0
Numato Opsis in 'operational' mode at /dev/bus/usb/001/008
 /sys/bus/usb/devices/1-1.2.2
 /sys/bus/usb/devices/1-1.2.2:1.0
 /sys/bus/usb/devices/1-1.2.2:1.1
 /sys/bus/usb/devices/1-1.2.2:1.2
 /sys/bus/usb/devices/1-1.2.2:1.3
 Board is currently used by drivers ('cdc_acm', 'uvcvideo')
 Serial port at /dev/ttyACM1
 Ignore as not type atlys
My root dir: /home/pi/HDMI2USB-mode-switch/venv/lib/python3.5/site-packages/hdmi2usb.modeswitch-0.0.1.post39-py3.5.egg/hdmi2usb/modeswitch
 Found exart-uarts at [LsusbDevice(04e2:1410:0003 /dev/bus/usb/001/015)] associating with Atlys at [Board(dev=LsusbDevice(1443:0007:0000 /dev/bus/usb/001/014), type='atlys', state='unconfigured')]
Digilent Atlys in 'unconfigured' mode at /dev/bus/usb/001/014
 /sys/bus/usb/devices/1-1.3.1
 /sys/bus/usb/devices/1-1.3.1:1.0
 Serial port at /dev/ttyVIZ0
Numato Opsis in 'operational' mode at /dev/bus/usb/001/008
 /sys/bus/usb/devices/1-1.2.2
 /sys/bus/usb/devices/1-1.2.2:1.0
 /sys/bus/usb/devices/1-1.2.2:1.1
 /sys/bus/usb/devices/1-1.2.2:1.2
 /sys/bus/usb/devices/1-1.2.2:1.3
 Board is currently used by drivers ('cdc_acm', 'uvcvideo')
 Serial port at /dev/ttyACM1
 Ignore as not type atlys
Using FX2 firmware hdmi2usb.hex
Running 'fxload -t fx2lp -D /dev/bus/usb/001/014 -I /home/pi/prebuilt/hdmi2usb.hex -v'
 Found exart-uarts at [LsusbDevice(04e2:1410:0003 /dev/bus/usb/001/015)] associating with Atlys at []
Traceback (most recent call last):
  File "/home/pi/HDMI2USB-mode-switch/venv/bin/hdmi2usb-mode-switch", line 11, in <module>
    load_entry_point('hdmi2usb.modeswitch==0.0.1.post39', 'console_scripts', 'hdmi2usb-mode-switch')()
  File "/home/pi/HDMI2USB-mode-switch/venv/lib/python3.5/site-packages/hdmi2usb.modeswitch-0.0.1.post39-py3.5.egg/hdmi2usb/modeswitch/cli.py", line 368, in main
    found_boards = find_boards(args)
  File "/home/pi/HDMI2USB-mode-switch/venv/lib/python3.5/site-packages/hdmi2usb.modeswitch-0.0.1.post39-py3.5.egg/hdmi2usb/modeswitch/cli.py", line 171, in find_boards
    all_boards = boards.find_boards(verbose=args.verbose)
  File "/home/pi/HDMI2USB-mode-switch/venv/lib/python3.5/site-packages/hdmi2usb.modeswitch-0.0.1.post39-py3.5.egg/hdmi2usb/modeswitch/boards.py", line 525, in find_boards
    assert len(exart_uarts) == len(atlys_boards)
AssertionError
```

run again, no error:

```
sometimes it doesn't error:
```
pi@oppi:~/prebuilt $ ~/HDMI2USB-mode-switch/venv/bin/hdmi2usb-mode-switch -v --by-type atlys --load-fx2-firmware hdmi2usb.hex
WARNING:root:unbind-helper not found, will have to run as root!
 Found exart-uarts at [LsusbDevice(04e2:1410:0003 /dev/bus/usb/001/010)] associating with Atlys at [Board(dev=LsusbDevice(1d50:60b7:0002 '0123456789abcdef'), type='atlys', state='operational')]
Digilent Atlys in 'operational' mode at /dev/bus/usb/001/012
 /sys/bus/usb/devices/1-1.3.1
 /sys/bus/usb/devices/1-1.3.1:1.0
 /sys/bus/usb/devices/1-1.3.1:1.1
 /sys/bus/usb/devices/1-1.3.1:1.2
 /sys/bus/usb/devices/1-1.3.1:1.3
 Board is currently used by drivers ('cdc_acm', 'uvcvideo')
 Serial port at /dev/ttyVIZ0, /dev/ttyACM2
Numato Opsis in 'operational' mode at /dev/bus/usb/001/008
 /sys/bus/usb/devices/1-1.2.2
 /sys/bus/usb/devices/1-1.2.2:1.0
 /sys/bus/usb/devices/1-1.2.2:1.1
 /sys/bus/usb/devices/1-1.2.2:1.2
 /sys/bus/usb/devices/1-1.2.2:1.3
 Board is currently used by drivers ('cdc_acm', 'uvcvideo')
 Serial port at /dev/ttyACM1
 Ignore as not type atlys
My root dir: /home/pi/HDMI2USB-mode-switch/venv/lib/python3.5/site-packages/hdmi2usb.modeswitch-0.0.1.post39-py3.5.egg/hdmi2usb/modeswitch
 Found exart-uarts at [LsusbDevice(04e2:1410:0003 /dev/bus/usb/001/010)] associating with Atlys at [Board(dev=LsusbDevice(1d50:60b7:0002 '0123456789abcdef'), type='atlys', state='operational')]
Digilent Atlys in 'operational' mode at /dev/bus/usb/001/012
 /sys/bus/usb/devices/1-1.3.1
 /sys/bus/usb/devices/1-1.3.1:1.0
 /sys/bus/usb/devices/1-1.3.1:1.1
 /sys/bus/usb/devices/1-1.3.1:1.2
 /sys/bus/usb/devices/1-1.3.1:1.3
 Board is currently used by drivers ('cdc_acm', 'uvcvideo')
 Serial port at /dev/ttyVIZ0, /dev/ttyACM2
Numato Opsis in 'operational' mode at /dev/bus/usb/001/008
 /sys/bus/usb/devices/1-1.2.2
 /sys/bus/usb/devices/1-1.2.2:1.0
 /sys/bus/usb/devices/1-1.2.2:1.1
 /sys/bus/usb/devices/1-1.2.2:1.2
 /sys/bus/usb/devices/1-1.2.2:1.3
 Board is currently used by drivers ('cdc_acm', 'uvcvideo')
 Serial port at /dev/ttyACM1
 Ignore as not type atlys
Detaching drivers from board.
Using FX2 firmware hdmi2usb.hex
Running 'fxload -t fx2lp -D /dev/bus/usb/001/012 -I /home/pi/prebuilt/hdmi2usb.hex -v'
 Found exart-uarts at [LsusbDevice(04e2:1410:0003 /dev/bus/usb/001/010)] associating with Atlys at [Board(dev=LsusbDevice(1d50:60b7:0002 '0123456789abcdef'), type='atlys', state='operational')]
Digilent Atlys in 'operational' mode at /dev/bus/usb/001/013
 /sys/bus/usb/devices/1-1.3.1
 /sys/bus/usb/devices/1-1.3.1:1.0
 /sys/bus/usb/devices/1-1.3.1:1.1
 /sys/bus/usb/devices/1-1.3.1:1.2
 /sys/bus/usb/devices/1-1.3.1:1.3
 Board is currently used by drivers ('cdc_acm', 'uvcvideo')
 Serial port at /dev/ttyVIZ0, /dev/ttyACM2
Numato Opsis in 'operational' mode at /dev/bus/usb/001/008
 /sys/bus/usb/devices/1-1.2.2
 /sys/bus/usb/devices/1-1.2.2:1.0
 /sys/bus/usb/devices/1-1.2.2:1.1
 /sys/bus/usb/devices/1-1.2.2:1.2
 /sys/bus/usb/devices/1-1.2.2:1.3
 Board is currently used by drivers ('cdc_acm', 'uvcvideo')
 Serial port at /dev/ttyACM1
 Ignore as not type atlys
Found 1 boards.
```

mithro, 2019-06-27:
This has both an Opsis and an Atlys plugged into the same machine?

CarlFK, 2019-06-28:
Yes.

both powered on.  

I am guessing "something" (mysterious!) is changing the state of .. something.  

Much like when --by-type opsis --flash-image foo.bin, it is fairly predictable it will take 3 tries:
with this pattern of DNA:
```
DNA = 110000001100000011000000110000001100000011000000110000001 (0x181818181818181)
DNA = 110000001100000011000000110000001100000010011011011000010 (0x1818181818136c2)
DNA = 100110110110000100011001100011000111100000011111101011000 (0x136c23318f03f58)
```
(I am now looking at the diffs of the 3 runs)

mithro, 2019-06-28:
Having both plugged into the same machine will make it very hard to debug.

CarlFK, 2019-06-28:
3 worked, 4th errored.

pi@oppi:~/prebuilt $ opsis-off.sh 
(click)
pi@oppi:~/prebuilt $ ~/HDMI2USB-mode-switch/venv/bin/hdmi2usb-mode-switch -v --by-type atlys --load-fx2-firmware hdmi2usb.hex
WARNING:root:unbind-helper not found, will have to run as root!
 Found exart-uarts at [LsusbDevice(04e2:1410:0003 /dev/bus/usb/001/007)] associating with Atlys at [Board(dev=LsusbDevice(1d50:60b7:0002 '0123456789abcdef'), type='atlys', state='operational')]
Digilent Atlys in 'operational' mode at /dev/bus/usb/001/010
 /sys/bus/usb/devices/1-1.3.4
 /sys/bus/usb/devices/1-1.3.4:1.0
 /sys/bus/usb/devices/1-1.3.4:1.1
 /sys/bus/usb/devices/1-1.3.4:1.2
 /sys/bus/usb/devices/1-1.3.4:1.3
 Board is currently used by drivers ('uvcvideo', 'cdc_acm')
 Serial port at /dev/ttyVIZ0, /dev/ttyACM2
My root dir: /home/pi/HDMI2USB-mode-switch/venv/lib/python3.5/site-packages/hdmi2usb.modeswitch-0.0.1.post39-py3.5.egg/hdmi2usb/modeswitch
 Found exart-uarts at [LsusbDevice(04e2:1410:0003 /dev/bus/usb/001/007)] associating with Atlys at [Board(dev=LsusbDevice(1d50:60b7:0002 '0123456789abcdef'), type='atlys', state='operational')]
Digilent Atlys in 'operational' mode at /dev/bus/usb/001/010
 /sys/bus/usb/devices/1-1.3.4
 /sys/bus/usb/devices/1-1.3.4:1.0
 /sys/bus/usb/devices/1-1.3.4:1.1
 /sys/bus/usb/devices/1-1.3.4:1.2
 /sys/bus/usb/devices/1-1.3.4:1.3
 Board is currently used by drivers ('uvcvideo', 'cdc_acm')
 Serial port at /dev/ttyVIZ0, /dev/ttyACM2
Detaching drivers from board.
Using FX2 firmware hdmi2usb.hex
Running 'fxload -t fx2lp -D /dev/bus/usb/001/010 -I /home/pi/prebuilt/hdmi2usb.hex -v'
 Found exart-uarts at [LsusbDevice(04e2:1410:0003 /dev/bus/usb/001/007)] associating with Atlys at [Board(dev=LsusbDevice(1d50:60b7:0002 '0123456789abcdef'), type='atlys', state='operational')]
Digilent Atlys in 'operational' mode at /dev/bus/usb/001/013
 /sys/bus/usb/devices/1-1.3.4
 /sys/bus/usb/devices/1-1.3.4:1.0
 /sys/bus/usb/devices/1-1.3.4:1.1
 /sys/bus/usb/devices/1-1.3.4:1.2
 /sys/bus/usb/devices/1-1.3.4:1.3
 Board is currently used by drivers ('uvcvideo', 'cdc_acm')
 Serial port at /dev/ttyVIZ0, /dev/ttyACM0
Found 1 boards.
pi@oppi:~/prebuilt $ ~/HDMI2USB-mode-switch/venv/bin/hdmi2usb-mode-switch -v --by-type atlys --load-fx2-firmware hdmi2usb.hex
WARNING:root:unbind-helper not found, will have to run as root!
 Found exart-uarts at [LsusbDevice(04e2:1410:0003 /dev/bus/usb/001/007)] associating with Atlys at [Board(dev=LsusbDevice(1d50:60b7:0002 '0123456789abcdef'), type='atlys', state='operational')]
Digilent Atlys in 'operational' mode at /dev/bus/usb/001/013
 /sys/bus/usb/devices/1-1.3.4
 /sys/bus/usb/devices/1-1.3.4:1.0
 /sys/bus/usb/devices/1-1.3.4:1.1
 /sys/bus/usb/devices/1-1.3.4:1.2
 /sys/bus/usb/devices/1-1.3.4:1.3
 Board is currently used by drivers ('cdc_acm', 'uvcvideo')
 Serial port at /dev/ttyVIZ0, /dev/ttyACM0
My root dir: /home/pi/HDMI2USB-mode-switch/venv/lib/python3.5/site-packages/hdmi2usb.modeswitch-0.0.1.post39-py3.5.egg/hdmi2usb/modeswitch
 Found exart-uarts at [LsusbDevice(04e2:1410:0003 /dev/bus/usb/001/007)] associating with Atlys at [Board(dev=LsusbDevice(1d50:60b7:0002 '0123456789abcdef'), type='atlys', state='operational')]
Digilent Atlys in 'operational' mode at /dev/bus/usb/001/013
 /sys/bus/usb/devices/1-1.3.4
 /sys/bus/usb/devices/1-1.3.4:1.0
 /sys/bus/usb/devices/1-1.3.4:1.1
 /sys/bus/usb/devices/1-1.3.4:1.2
 /sys/bus/usb/devices/1-1.3.4:1.3
 Board is currently used by drivers ('cdc_acm', 'uvcvideo')
 Serial port at /dev/ttyVIZ0, /dev/ttyACM0
Detaching drivers from board.
Using FX2 firmware hdmi2usb.hex
Running 'fxload -t fx2lp -D /dev/bus/usb/001/013 -I /home/pi/prebuilt/hdmi2usb.hex -v'
 Found exart-uarts at [LsusbDevice(04e2:1410:0003 /dev/bus/usb/001/007)] associating with Atlys at [Board(dev=LsusbDevice(1d50:60b7:0002 '0123456789abcdef'), type='atlys', state='operational')]
Digilent Atlys in 'operational' mode at /dev/bus/usb/001/014
 /sys/bus/usb/devices/1-1.3.4
 /sys/bus/usb/devices/1-1.3.4:1.0
 /sys/bus/usb/devices/1-1.3.4:1.1
 /sys/bus/usb/devices/1-1.3.4:1.2
 /sys/bus/usb/devices/1-1.3.4:1.3
 Board is currently used by drivers ('cdc_acm', 'uvcvideo')
[... 45 lines cut ...]
 Board is currently used by drivers ('cdc_acm', 'uvcvideo')
 Serial port at /dev/ttyVIZ0, /dev/ttyACM0
My root dir: /home/pi/HDMI2USB-mode-switch/venv/lib/python3.5/site-packages/hdmi2usb.modeswitch-0.0.1.post39-py3.5.egg/hdmi2usb/modeswitch
 Found exart-uarts at [LsusbDevice(04e2:1410:0003 /dev/bus/usb/001/007)] associating with Atlys at [Board(dev=LsusbDevice(1d50:60b7:0002 '0123456789abcdef'), type='atlys', state='operational')]
Digilent Atlys in 'operational' mode at /dev/bus/usb/001/015
 /sys/bus/usb/devices/1-1.3.4
 /sys/bus/usb/devices/1-1.3.4:1.0
 /sys/bus/usb/devices/1-1.3.4:1.1
 /sys/bus/usb/devices/1-1.3.4:1.2
 /sys/bus/usb/devices/1-1.3.4:1.3
 Board is currently used by drivers ('cdc_acm', 'uvcvideo')
 Serial port at /dev/ttyVIZ0, /dev/ttyACM0
Detaching drivers from board.
Using FX2 firmware hdmi2usb.hex
Running 'fxload -t fx2lp -D /dev/bus/usb/001/015 -I /home/pi/prebuilt/hdmi2usb.hex -v'
 Found exart-uarts at [LsusbDevice(04e2:1410:0003 /dev/bus/usb/001/007)] associating with Atlys at []
Traceback (most recent call last):
  File "/home/pi/HDMI2USB-mode-switch/venv/bin/hdmi2usb-mode-switch", line 11, in <module>
    load_entry_point('hdmi2usb.modeswitch==0.0.1.post39', 'console_scripts', 'hdmi2usb-mode-switch')()
[... cut ...] line 368, in main
    found_boards = find_boards(args)
  File "/home/pi/HDMI2USB-mode-switch/venv/lib/python3.5/site-packages/hdmi2usb.modeswitch-0.0.1.post39-py3.5.egg/hdmi2usb/modeswitch/cli.py", line 171, in find_boards
    all_boards = boards.find_boards(verbose=args.verbose)
  File "/home/pi/HDMI2USB-mode-switch/venv/lib/python3.5/site-packages/hdmi2usb.modeswitch-0.0.1.post39-py3.5.egg/hdmi2usb/modeswitch/boards.py", line 525, in find_boards
    assert len(exart_uarts) == len(atlys_boards)

mithro, 2019-06-28:
Change the following line
```python
    assert len(exart_uarts) == len(atlys_boards)
```
to
```python
    assert len(exart_uarts) == len(atlys_boards), repr((exart_uarts, atlys_boards))
```
Found in `/home/pi/HDMI2USB-mode-switch/venv/lib/python3.5/site-packages/hdmi2usb.modeswitch-0.0.1.post39-py3.5.egg/hdmi2usb/modeswitch/boards.py` - line 525

CarlFK, 2019-06-28:
AssertionError: ([LsusbDevice(04e2:1410:0003 /dev/bus/usb/001/021)], [])

```
pi@oppi:~/prebuilt $ ~/HDMI2USB-mode-switch/venv/bin/hdmi2usb-mode-switch -v --by-type atlys --load-fx2-firmware hdmi2usb.hex
WARNING:root:unbind-helper not found, will have to run as root!
 Found exart-uarts at [LsusbDevice(04e2:1410:0003 /dev/bus/usb/001/021)] associating with Atlys at [Board(dev=LsusbDevice(1443:0007:0000 /dev/bus/usb/001/022), type='atlys', state='unconfigured')]
Digilent Atlys in 'unconfigured' mode at /dev/bus/usb/001/022
 /sys/bus/usb/devices/1-1.3.4
 /sys/bus/usb/devices/1-1.3.4:1.0
 Serial port at /dev/ttyVIZ0
My root dir: /home/pi/HDMI2USB-mode-switch/venv/lib/python3.5/site-packages/hdmi2usb.modeswitch-0.0.1.post39-py3.5.egg/hdmi2usb/modeswitch
 Found exart-uarts at [LsusbDevice(04e2:1410:0003 /dev/bus/usb/001/021)] associating with Atlys at [Board(dev=LsusbDevice(1443:0007:0000 /dev/bus/usb/001/022), type='atlys', state='unconfigured')]
Digilent Atlys in 'unconfigured' mode at /dev/bus/usb/001/022
 /sys/bus/usb/devices/1-1.3.4
 /sys/bus/usb/devices/1-1.3.4:1.0
 Serial port at /dev/ttyVIZ0
Using FX2 firmware hdmi2usb.hex
Running 'fxload -t fx2lp -D /dev/bus/usb/001/022 -I /home/pi/prebuilt/hdmi2usb.hex -v'
 Found exart-uarts at [LsusbDevice(04e2:1410:0003 /dev/bus/usb/001/021)] associating with Atlys at []
Traceback (most recent call last):
  File "/home/pi/HDMI2USB-mode-switch/venv/bin/hdmi2usb-mode-switch", line 11, in <module>
    load_entry_point('hdmi2usb.modeswitch==0.0.1.post39', 'console_scripts', 'hdmi2usb-mode-switch')()
  File "/home/pi/HDMI2USB-mode-switch/venv/lib/python3.5/site-packages/hdmi2usb.modeswitch-0.0.1.post39-py3.5.egg/hdmi2usb/modeswitch/cli.py", line 368, in main
    found_boards = find_boards(args)
  File "/home/pi/HDMI2USB-mode-switch/venv/lib/python3.5/site-packages/hdmi2usb.modeswitch-0.0.1.post39-py3.5.egg/hdmi2usb/modeswitch/cli.py", line 171, in find_boards
    all_boards = boards.find_boards(verbose=args.verbose)
  File "/home/pi/HDMI2USB-mode-switch/venv/lib/python3.5/site-packages/hdmi2usb.modeswitch-0.0.1.post39-py3.5.egg/hdmi2usb/modeswitch/boards.py", line 525, in find_boards
    assert len(exart_uarts) == len(atlys_boards), repr((exart_uarts, atlys_boards))
AssertionError: ([LsusbDevice(04e2:1410:0003 /dev/bus/usb/001/021)], [])
```

mithro, 2019-06-28:
Give https://github.com/mithro/HDMI2USB-mode-switch/commit/2a5f659cc0ff3e2631a5b4546dbdd3a95e2cb202 a go?

CarlFK, 2019-06-28:
5 or 10 times now, no error

it is in a loop until I want to do something else and I'm convinced the bug is fixed.

looks like it has run 10 - 20 times now, no error, so prolly fixed.

mithro, 2019-07-02:
I'm saying this is fixed now then.

closed 2019-07-02 by mithro